jaheyns/CfdOF #186: Bug reporting  'PySide6.QtCore.QProcess' object has no attribute 'Timedout'

yueduz, 2024-10-09:
After clicking the "install Docker Container" button
[image]
[image]

oliveroxtoby, 2024-10-10:
@mmcker perhaps you could have a look?

mmcker, 2024-10-10:
I see that arch linux uses freecad 1.0rc2-2 (as shown in the screenshot above)

It seems to be a freecad pyside wrapper / pyside6 issue.

A simple check from a debian system shows the attribute should be there - see below.  I tested a fresh install with the stable verison of freecad and it works correctly.  I'll install the 1.0rc2-2 (on debian) and see if I can see what is going on.

In the meantime, can you try the stable appimage?  

Python 3.13.0 | packaged by conda-forge | (main, Oct  8 2024, 20:04:32) [GCC 13.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from PySide6.QtCore import QProcess
>>> print(QProcess.Timedout)
ProcessError.Timedout
>>> print(QProcess.Timedout.value)
2

mmcker, 2024-10-10:
Hi yueduz

I just downloaded freecad 1.0rc2-2 appimage from here:

https://github.com/FreeCAD/FreeCAD/releases/tag/1.0rc2

And everything seems to work (I deleted the docker image and then reinstalled it through the cfdof addon).

It must be an issue with the arch linux native package for freecad.  I suggest uninstalling it and using the appimage download from the above link.

I believe you will need to install the fuse2 package to run the appimage.

yueduz, 2024-10-11:
After using freecad 1.0rc2-2 appimage, it works fine

closed 2024-10-11 by yueduz