jaheyns/CfdOF #190: CfdOF causes BooleanFragments containers to switch mode to CompSolid on document load

AIRCAP, 2024-10-28:
[CFDOfbroken.zip](https://github.com/user-attachments/files/17547055/CFDOfbroken.zip)

[image]

In this file, when CfdOF is installed unless the (empty) CfdOF Analysis container is removed, on loading the document the BooleanFragments feature changes mode to "CompSolid"

Without CfdOF installed or without the Analysis container, the feature remains Standard as it should.

This is a distilled test case - removing things from the file broke a reference in the analysis container, so it currently fails recomputation due to missing ref to a spreadsheet, but the weird side effect was also there when the analysis container was fully functional (I can't share that file)

The issue with CompSolid is that fragments belonging to compounds initially used in the BooleanFragment operation are no longer grouped together if CompSolid is exploded. This is however needed for assigning materials in the FEM workbench and other operations.
CfdOF should not change or affect properties of features that are not even referenced in the CfdOF Analysis container.

**Steps to reproduce:**

1. Load the file
2. Change the mode on the BooleanFragments feature to "Standard" if it isn't
3. Save the file and close it
4. Load the file again with CfdOF installed

**Expected outcome:**
Mode is Standard

**Actual Outcome:**
Mode switches to CompSolid and the BooleanFragments container is marked for recomputation

This issue has been previously reported on FreeCAD as 
https://github.com/FreeCAD/FreeCAD/issues/17533

oliveroxtoby, 2024-10-28:
Sorry about that @AIRCAP. This was some very ancient and dodgy code that really shouldn't have been there. I've removed it and hopefully all will be well now.

closed 2024-10-28 by oliveroxtoby

AIRCAP, 2024-10-28:
Thanks for the quick fix :-)

commit 280bf700 (2024-10-28): Disable changing parts indiscriminately to CompSolid mode

This was causing interoperability issues.

Resolves #190