jaheyns/CfdOF #196: "No module named 'PySide.QtWidgets'" occurred in /home/usr/.FreeCAD/Mod/CfdOF/InitGui.py

xphoniex, 2025-01-25:
```
OS: Ubuntu 22.04 LTS (bspwm)
Word size of OS: 64-bit
Word size of FreeCAD: 64-bit
Version: 0.19.
Build type: Release
Python version: 3.10.2
Qt version: 5.15.2
Coin version: 4.0.0
OCC version: 7.5.1
Locale: English/United States (en_US)
```

Tried to install CfdOF via addon manager, sucessfully gets installed but on restart there's no workbench and there's error in the **Report View**:

```
20:16:56  During initialization the error "No module named 'PySide.QtWidgets'" occurred in /home/da/.FreeCAD/Mod/CfdOF/InitGui.py
20:16:56  Please look into the log file for further information
```

xphoniex, 2025-01-25:
Followed [this](https://github.com/FreeCAD/FreeCAD-render/issues/415#issuecomment-2195683109) and now I'm getting:

```
20:36:42  During initialization the error "No module named 'CfdOF.compat'" occurred in /home/usr/.FreeCAD/Mod/CfdOF/InitGui.py
20:36:42  Please look into the log file for further information
```

xphoniex, 2025-01-25:
Can't find neither `compat` nor `Plot` anywhere in the repo, so installed Plot workspace and then changed **.FreeCAD/Mod/CfdOF/CfdOF/CfdTimePlot.py** code to:

```python
#if int(FreeCAD.Version()[0]) == 0 and int(FreeCAD.Version()[1].split('.')[0]) < 20:
if int(FreeCAD.Version()[0]) == 0 and int(FreeCAD.Version()[1].split('.')[0]) > 20:
    from CfdOF.compat import Plot  # Plot workbench
else:
    try: 
        #from FreeCAD.Plot import Plot  # Inbuilt plot module
        from FreeCAD import Plot  # Inbuilt plot module

    except ImportError:
        from CfdOF.compat import Plot  # Fallback to compat (should be unnecessary once 0.20 is stable)
```

oliveroxtoby, 2025-01-27:
> ```
> Version: 0.19.
> ```

I'm afraid this version is too old for CfdOF. Please could you try again using the latest version of FreeCAD?

closed 2025-01-27 by oliveroxtoby