jaheyns/CfdOF #197: Unable to install cfmesh/HISA

ripjohnbrown1859, 2025-01-26:
I have been trying for 3 days to get freecad to work with cfdof. I have successfully installed openfoam and paraview, both of which i had to do manually, but whenever i try to install cfmesh or hisa the download takes an extremely long time and then either crashed or reports installing successfully and is not recognized by the dependency check. Is there a way to manually install them? I cannot find any output log files in the target output directory either.

ripjohnbrown1859, 2025-01-26:
[image]

[image]

oliveroxtoby, 2025-01-27:
Strange - I haven't seen it do this before. 
As a workaround, what you can do is download the files manually using the URLs displayed in the "URL:" text boxes and then use the "Choose Existing File..." buttons to point CfdOF to those downloaded files on your computer. It should then install from the local file when you choose "Install cfMesh/HiSA".

ripjohnbrown1859, 2025-01-27:
> Strange - I haven't seen it do this before. As a workaround, what you can do is download the files manually using the URLs displayed in the "URL:" text boxes and then use the "Choose Existing [...]
 when i do that it does the same thing.

ripjohnbrown1859, 2025-01-27:
i have since completely reinstalled it on WSL2 now and its doing the same exact thing

ripjohnbrown1859, 2025-01-27:
[image]

oliveroxtoby, 2025-01-30:
> i have since completely reinstalled it on WSL2 now and its doing the same exact thing

WSL is unfortunately not supported currently. 

I can only suggest that you try running FreeCAD as administrator, and if that doesn't work try re-installing OpenFOAM somewhere outside the Program Files directory, so that there is no issue with write permissions to the folder. 

The Forum is probably the place for futher discussion on this: https://forum.freecad.org/viewforum.php?f=37&sid=ba1670bb829341965e32fa407b022e17

The following introductory post may be of some use: https://forum.freecad.org/viewtopic.php?f=37&t=33492

Michkovy, 2025-05-04:
I had similar problems to ripjpohnbrown1895. I was following the [readme ](https://github.com/jaheyns/CfdOF/tree/master?tab=readme-ov-file#windows-1) and something that isn't mentioned is that you need to match the OF version to HiSA and gmsh executables. Since I got bamboozled between the two OF installers, I used the FC settings menu to download one. That points to v2212. The other dependencies point to the latest version, which are not necessarily v2212 compatible anymore.