jaheyns/CfdOF #198: Paraview crashing only when launching through CFDOF

Digitalpeer1, 2025-02-25:
Hi,

I am using this extension to run some simulations but when I hit the ParaView button to launch it it immediately crashes as seen in the video attached. Paraview launches when clicked on task bar but when I launch it from Freecad it crashes. I am using Freecad 1.0 app image on fedora 41 KDE. Please let me know If there are any log files that I can attach I am not aware of any at this moment.

https://github.com/user-attachments/assets/e86e30bc-d0db-4e84-9cb5-d1fd6576297f

oliveroxtoby, 2025-03-05:
The Appimage modifies the environment in which programs are run, which causes problems trying to run external programs in the system environment. CfdOF tries to compensate for that, but there is most likely something on your system that needs to be accounted for in this logic.

To help debug, could you please go to Edit | Preferences, CfdOF, then click 'Run dependency checker', then wait for it to complete. Close the Preferences window, and paste the contents of the Report View here.

Also, please type in the python console:
```
import os
os.env
```
and paste the results here.

Finally, could you go to a terminal and type 'export' and paste that output here.

Thanks

Ruscios, 2025-04-08:
Having the same issue, Fedora 41 KDE, running AppImage version of FreeCAD 1.0. 

Dependency checker reports:
`Checking dependencies...
FreeCAD version: 1.0
System: Linux
Runtime: Posix
OpenFOAM directory: /usr/lib/openfoam/openfoam2412
OpenFOAM version: 2412
OpenFOAM version 2412 is not yet supported:
Last tested version is 2312
OpenFOAM installation does not include 'wmake'. Installation of cfMesh and HiSA will not be possible. An OpenFOAM 'development' package should be installed if available.
cfMesh (CfdOF version) not found
HiSA not found
Paraview executable: /usr/bin/paraview
Paraview version: 5.12.1
gmsh executable: /tmp/.mount_freecaNmhbaM/usr/bin/gmsh
gmsh version: 4.12.2
Completed CFD dependency check`

Python commands return:
`Traceback (most recent call last):
  File "<input>", line 1, in <module>
AttributeError: module 'os' has no attribute 'env'`

Export returns:
`declare -x BASH_ENV="/usr/share/lmod/lmod/init/bash"
declare -x COLORFGBG="15;0"
declare -x COLORTERM="truecolor"
declare -x DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"
declare -x DEBUGINFOD_IMA_CERT_PATH="/etc/keys/ima:"
declare -x DEBUGINFOD_URLS="https://debuginfod.fedoraproject.org/ "
declare -x DESKTOP_SESSION="plasma"
declare -x DISPLAY=":0"
declare -x EDITOR="/usr/bin/nano"
declare -x FPATH="/usr/share/lmod/lmod/init/ksh_funcs"
declare -x GDK_CORE_DEVICE_EVENTS="1"
declare -x GPG_TTY="/dev/pts/2"
declare -x GTK2_RC_FILES="/home/todd/.gtkrc-2.0-kde4"
declare -x GTK_RC_FILES="/etc/gtk/gtkrc:/home/todd/.gtkrc:/home/todd/.config/gtkrc"
declare -x HISTCONTROL="ignoredups"
declare -x HISTSIZE="1000"
declare -x HOME="/home/todd"
declare -x HOSTNAME="Theseus"
declare -x ICEAUTHORITY="/run/user/1000/iceauth_PDymwv"
declare -x IMSETTINGS_INTEGRATE_DESKTOP="yes"
declare -x IMSETTINGS_MODULE="none"
declare -x INVOCATION_ID="d1ba327266fb4e2eb8d0dcbcb1d73ed1"
declare -x JOURNAL_STREAM="9:35886"
declare -x KDEDIRS="/usr"
declare -x KDE_APPLICATIONS_AS_SCOPE="1"
declare -x KDE_FULL_SESSION="true"
declare -x KDE_SESSION_UID="1000"
declare -x KDE_SESSION_VERSION="6"
declare -x KONSOLE_DBUS_SERVICE=":1.265"
declare -x KONSOLE_DBUS_SESSION="/Sessions/2"
declare -x KONSOLE_DBUS_WINDOW="/Windows/1"
declare -x KONSOLE_VERSION="241203"
declare -x LANG="en_US.UTF-8"
declare -x LANGUAGE=""
declare -x LESSOPEN="||/usr/bin/lesspipe.sh %s"
declare -x LMOD_CMD="/usr/share/lmod/lmod/libexec/lmod"
declare -x LMOD_DIR="/usr/share/lmod/lmod/libexec"
declare -x LMOD_PKG="/usr/share/lmod/lmod"
declare -x LMOD_ROOT="/usr/share/lmod"
declare -x LMOD_SETTARG_FULL_SUPPORT="no"
declare -x LMOD_VERSION="8.7.55"
declare -x LMOD_sys="Linux"
declare -x LOGNAME="todd"
declare -x LS_COLORS="rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=01;37;41:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.7z=01;31:*.ace=01;31:*.alz=01;31:*.apk=01;31:*.arc=01;31:*.arj=01;31:*.bz=01;31:*.bz2=01;31:*.cab=01;31:*.cpio=01;31:*.crate=01;31:*.deb=01;31:*.drpm=01;31:*.dwm=01;31:*.dz=01;31:*.ear=01;31:*.egg=01;31:*.esd=01;31:*.gz=01;31:*.jar=01;31:*.lha=01;31:*.lrz=01;31:*.lz=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.lzo=01;31:*.pyz=01;31:*.rar=01;31:*.rpm=01;31:*.rz=01;31:*.sar=01;31:*.swm=01;31:*.t7z=01;31:*.tar=01;31:*.taz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tgz=01;31:*.tlz=01;31:*.txz=01;31:*.tz=01;31:*.tzo=01;31:*.tzst=01;31:*.udeb=01;31:*.war=01;31:*.whl=01;31:*.wim=01;31:*.xz=01;31:*.z=01;31:*.zip=01;31:*.zoo=01;31:*.zst=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=01;36:*.au=01;36:*.flac=01;36:*.m4a=01;36:*.mid=01;36:*.midi=01;36:*.mka=01;36:*.mp3=01;36:*.mpc=01;36:*.ogg=01;36:*.ra=01;36:*.wav=01;36:*.oga=01;36:*.opus=01;36:*.spx=01;36:*.xspf=01;36:*~=00;90:*#=00;90:*.bak=00;90:*.crdownload=00;90:*.dpkg-dist=00;90:*.dpkg-new=00;90:*.dpkg-old=00;90:*.dpkg-tmp=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:*.swp=00;90:*.tmp=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:"
[... 27 lines cut ...]
declare -x WINDOWID="1"
declare -x XAUTHORITY="/run/user/1000/xauth_brDvIQ"
declare -x XDG_CONFIG_DIRS="/home/todd/.config/kdedefaults:/etc/xdg:/usr/share/kde-settings/kde-profile/default/xdg"
declare -x XDG_CURRENT_DESKTOP="KDE"
declare -x XDG_DATA_DIRS="/home/todd/.local/share/flatpak/exports/share:/var/lib/flatpak/exports/share:/usr/local/share:/usr/share:/var/lib/snapd/desktop"
declare -x XDG_MENU_PREFIX="plasma-"
declare -x XDG_RUNTIME_DIR="/run/user/1000"
declare -x XDG_SEAT="seat0"
declare -x XDG_SEAT_PATH="/org/freedesktop/DisplayManager/Seat0"
declare -x XDG_SESSION_CLASS="user"
declare -x XDG_SESSION_DESKTOP="KDE"
declare -x XDG_SESSION_ID="2"
declare -x XDG_SESSION_PATH="/org/freedesktop/DisplayManager/Session1"
declare -x XDG_SESSION_TYPE="wayland"
declare -x XDG_VTNR="2"
declare -x XKB_DEFAULT_LAYOUT="us"
declare -x XKB_DEFAULT_MODEL="pc105"
declare -x XMODIFIERS="@im=none"
`

Additionally, if I run paraview from terminal with the script argument like this:
`/usr/bin/paraview --script=/tmp/case/pvScript.py`
It returns:
`Qt: Session management error: Could not open network socket
Segmentation fault (core dumped)`

oliveroxtoby, 2025-04-08:
Thanks for gathering the data.

> 
> Python commands return:
> 
> ```
> Traceback (most recent call last):
>   File "<input>", line 1, in <module>
> AttributeError: module 'os' has no attribute 'env'
> ```

Sorry about this...it should have been
```
import os
os.environ
```

> Additionally, if I run paraview from terminal with the script argument like this: `/usr/bin/paraview --script=/tmp/case/pvScript.py` It returns:
> 
> ```
> Qt: Session management error: Could not open network socket
> Segmentation fault (core dumped)
> ```

Aha, this is useful. So just to confirm, if you run `/usr/bin/paraview` from the terminal, it opens successfully, and the problem only occurs when including the --script option?

Ruscios, 2025-04-09:
Yes, just when I use the script option. I've discovered that the Fedora 41 repository has the 5.12.1 version of Paraview, while the current ParaView version is 5.13.3. The 5.12.1 version, in both binary and flatpak form, crashes whenever the script editor is opened with nothing else even loaded. The 5.13.3 version lets you open the script editor, but I've only gotten it working as a flatpak on my system thus far. I've tried launching with the pvScript.py two ways, one in the default `/tmp/case` directory, and another where I've moved the case directory elesewhere.

Results of `/var/lib/flatpak/exports/bin/org.paraview.ParaView --script=/tmp/case/pvScript.py`:
```
Qt: Session management error: Could not open network socket
critical: In unknown, line 0
critical: Cannot open Python script specified: ' /tmp/case/pvScript.py '
```
Paraview stays running, however.

Results of `/var/lib/flatpak/exports/bin/org.paraview.ParaView --script=/home/[user]/Documents/case/pvScript.py`:
```
Qt: Session management error: Could not open network socket
(   3.265s) [paraview        ]  vtkOpenFOAMReader.cxx:11406  ERR| vtkPOpenFOAMReader (0x5597dbbb7700): Cannot open directory: ./constant/
(   3.268s) [paraview        ]       vtkExecutive.cxx:729    ERR| vtkPVCompositeDataPipeline (0x5597db89a900): Algorithm vtkPOpenFOAMReader (0x5597dbbb7700) returned failure for request: vtkInformation (0x5597dc017660)
  Debug: Off
  Modified Time: 424305
  Reference Count: 1
  Registered Events: (none)
  Request: REQUEST_INFORMATION
  ALGORITHM_AFTER_FORWARD: 1
  FORWARD_DIRECTION: 0

Traceback (most recent call last):
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 475, in __setattr__
    setter = getattr(self.__class__, name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'GeometryRepresentation' has no attribute 'ScalarOpacityUnitDistance'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 481, in __setattr__
    _bc.setattr(self, name, value)
  File "/app/lib/python3.12/site-packages/paraview/_backwardscompatibilityhelper.py", line 523, in setattr
    raise AttributeError()
AttributeError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 34, in <module>
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 488, in __setattr__
    raise AttributeError("Attribute %s does not exist. " % name +
AttributeError: Attribute ScalarOpacityUnitDistance does not exist.  This class does not allow addition of new attributes to avoid mistakes due to typos. Use add_attribute() if you really want to add this attribute.
```
Paraview also stays running in this case.

I've tried to use the python script editor in this version, but the drop-down menus aren't working properly within the script editor so I can't load anything.

Part of this seems to be a Fedora packaging issue and a mix of old and new Paraview issues, but I don't know that that's the whole story.

Just in case though, os.environ returns:
`environ({'IMSETTINGS_INTEGRATE_DESKTOP': 'yes', 'SHELL': '/bin/bash', 'SESSION_MANAGER': 'local/unix:@/tmp/.ICE-unix/15614,unix/unix:/tmp/.ICE-unix/15614', 'XDG_CONFIG_DIRS': '/home/todd/.config/kdedefaults:/etc/xdg:/usr/share/kde-settings/kde-profile/default/xdg', 'XDG_SESSION_PATH': '/org/freedesktop/DisplayManager/Session1', 'HISTCONTROL': 'ignoredups', 'XDG_MENU_PREFIX': 'plasma-', 'HOSTNAME': 'Theseus', 'HISTSIZE': '1000', 'ICEAUTHORITY': '/run/user/1000/iceauth_PDymwv', 'FONTCONFIG_PATH': '/etc/fonts', 'PREFIX': '/tmp/.mount_freecaNmhbaM/usr', 'FPATH': '/usr/share/lmod/lmod/init/ksh_funcs', 'SSH_AUTH_SOCK': '/run/user/1000/ssh-agent.socket', 'ARGV0': '/home/todd/AppImages/freecad.appimage', 'MEMORY_PRESSURE_WRITE': 'c29tZSAyMDAwMDAgMjAwMDAwMAA=', 'XMODIFIERS': '@im=none', 'DESKTOP_SESSION': 'plasma', 'GTK_RC_FILES': '/etc/gtk/gtkrc:/home/todd/.gtkrc:/home/todd/.config/gtkrc', 'GDK_CORE_DEVICE_EVENTS': '1', 'GPG_TTY': '/dev/tty2', 'LMOD_DIR': '/usr/share/lmod/lmod/libexec', 'EDITOR': '/usr/bin/nano', 'XDG_SEAT': 'seat0', 'PWD': '/home/todd', 'XDG_SESSION_DESKTOP': 'KDE', 'LOGNAME': 'todd', 'XDG_SESSION_TYPE': 'wayland', 'MODULESHOME': '/usr/share/lmod/lmod', 'MANPATH': '/usr/share/lmod/lmod/share/man:', 'SYSTEMD_EXEC_PID': '16321', 'XAUTHORITY': '/run/user/1000/xauth_brDvIQ', 'DESKTOP_STARTUP_ID': 'not-granted-666', 'XKB_DEFAULT_MODEL': 'pc105', 'GTK2_RC_FILES': '/home/todd/.gtkrc-2.0-kde4', 'HOME': '/home/todd', 'GIT_SSL_CAINFO': '/tmp/.mount_freecaNmhbaM/usr/ssl/cacert.pem', 'SSH_ASKPASS': '/usr/bin/ksshaskpass', 'LANG': 'en_US.UTF-8', 'LS_COLORS': '', 'XDG_CURRENT_DESKTOP': 'KDE', 'MEMORY_PRESSURE_WATCH': '/sys/fs/cgroup/user.slice/user-1000.slice/user@1000.service/app.slice/flatpak-session-helper.service/memory.pressure', 'WAYLAND_DISPLAY': 'wayland-0', 'LMOD_SETTARG_FULL_SUPPORT': 'no', 'XDG_SEAT_PATH': '/org/freedesktop/DisplayManager/Seat0', 'INVOCATION_ID': '731bfff3818d462ba9d473b91f7ae9cf', 'LMOD_VERSION': '8.7.55', 'MANAGERPID': '15040', 'IMSETTINGS_MODULE': 'none', 'APPDIR': '/tmp/.mount_freecaNmhbaM', 'KDE_SESSION_UID': '1000', 'MODULEPATH_ROOT': '/usr/share/modulefiles', 'MOZ_GMP_PATH': '/usr/lib64/mozilla/plugins/gmp-gmpopenh264/system-installed', 'XKB_DEFAULT_LAYOUT': 'us', 'XDG_ACTIVATION_TOKEN': 'not-granted-666', 'XDG_SESSION_CLASS': 'user', 'LMOD_PKG [... cut ...] _lmod_sh_dbg:-}" ]; then\n set +$__lmod_sh_dbg;\n echo "Shell debugging temporarily silenced: export LMOD_SH_DBG_ON=1 for Lmod\'s output" 1>&2;\n fi;\n eval "$($LMOD_CMD shell "$@")" && eval "$(${LMOD_SETTARG_CMD:-:} -s sh)";\n __lmod_my_status=$?;\n if [ -n "${__lmod_sh_dbg:-}" ]; then\n echo "Shell debugging restarted" 1>&2;\n set -$__lmod_sh_dbg;\n fi;\n unset __lmod_sh_dbg;\n return $__lmod_my_status\n}', '_': '/tmp/.mount_freecaNmhbaM/usr/bin/freecad', 'LC_NUMERIC': 'C', 'PYTHONPATH': ''})`

S0Phon, 2025-04-26:
I have the same issue and running Fedora 42. I found a workaround using the flatpak version of paraview and had to update the executable path to '/var/lib/flatpak/exports/bin/org.paraview.ParaView'.

Please let me know if I should post any log files. I did find that Fedora's problem reporting application captured paraview crashing with the error 'paraview killed by SIGSEGV'.

Digitalpeer1, 2025-05-09:
Hi need to get openfoam working on Fedora 42 first having problems with MPI when compiling and cant install precompiled cuz repos will let you know if this works

Digitalpeer1, 2025-05-14:
Hi this seems to work as a temp solution thanks!

Digitalpeer1, 2025-05-19:
When opening files I do run into a new issue when opening mesh or finished run in paraview I get the error
 `critical: In unknown, line 0
critical: Cannot open Python script specified: ' pvScriptMesh.py '
`

Digitalpeer1, 2025-06-21:
Tried that here is the result

ERROR: In vtkOpenFOAMReader.cxx, line 11424
vtkPOpenFOAMReader (0x5602f4225380): pv.foam contains no meshes.

ERROR: In vtkExecutive.cxx, line 729
vtkPVCompositeDataPipeline (0x5602f3ee33a0): Algorithm vtkPOpenFOAMReader (0x5602f4225380) returned failure for request: vtkInformation (0x5602f459ed50)
  Debug: Off
  Modified Time: 424319
  Reference Count: 1
  Registered Events: (none)
  Request: REQUEST_INFORMATION
  ALGORITHM_AFTER_FORWARD: 1
  FORWARD_DIRECTION: 0

Traceback (most recent call last):
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 475, in __setattr__
    setter = getattr(self.__class__, name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'GeometryRepresentation' has no attribute 'ScalarOpacityUnitDistance'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 481, in __setattr__
    _bc.setattr(self, name, value)
  File "/app/lib/python3.12/site-packages/paraview/_backwardscompatibilityhelper.py", line 523, in setattr
    raise AttributeError()
AttributeError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 34, in <module>
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 488, in __setattr__
    raise AttributeError("Attribute %s does not exist. " % name +
AttributeError: Attribute ScalarOpacityUnitDistance does not exist.  This class does not allow addition of new attributes to avoid mistakes due to typos. Use add_attribute() if you really want to add this attribute.

Qt: Session management error: None of the authentication protocols specified are supported
(   1.275s) [paraview        ]  vtkOpenFOAMReader.cxx:11424  ERR| vtkPOpenFOAMReader (0x5602f4225380): pv.foam contains no meshes.
(   1.282s) [paraview        ]       vtkExecutive.cxx:729    ERR| vtkPVCompositeDataPipeline (0x5602f3ee33a0): Algorithm vtkPOpenFOAMReader (0x5602f4225380) returned failure for request: vtkInformation (0x5602f459ed50)
  Debug: Off
  Modified Time: 424319
  Reference Count: 1
  Registered Events: (none)
  Request: REQUEST_INFORMATION
  ALGORITHM_AFTER_FORWARD: 1
  FORWARD_DIRECTION: 0

Traceback (most recent call last):
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 475, in __setattr__
    setter = getattr(self.__class__, name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: type object 'GeometryRepresentation' has no attribute 'ScalarOpacityUnitDistance'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 481, in __setattr__
    _bc.setattr(self, name, value)
  File "/app/lib/python3.12/site-packages/paraview/_backwardscompatibilityhelper.py", line 523, in setattr
    raise AttributeError()
AttributeError

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<string>", line 34, in <module>
  File "/app/lib/python3.12/site-packages/paraview/servermanager.py", line 488, in __setattr__
    raise AttributeError("Attribute %s does not exist. " % name +
AttributeError: Attribute ScalarOpacityUnitDistance does not exist.  This class does not allow addition of new attributes to avoid mistakes due to typos. Use add_attribute() if you really want to add this attribute.

Digitalpeer1, 2025-06-21:
oh my bad one sec

Digitalpeer1, 2025-06-21:
Seems like it works now thanks!

closed 2025-06-21 by Digitalpeer1