jaheyns/CfdOF #199: Bug reporting - Unable to change Analysis Tolerance

AndrewLindsay, 2025-03-22:
When I have was working through the tutorials provided for CfdOF, Tutorial 3 required me to change the tolerance of the solver.  When I tried to change the tolerance to 0.002 like in the tutorial, I could only type in 0's I could not type in 0.002 or 0.X, even where X was any number.
I am running Ubuntu 24.04 on Parallels desktop on a Mac M3.  I am using the ARM version of everything where possible.

[image]

oliveroxtoby, 2025-03-24:
I think we're a bit limited here by how FreeCAD format numbers in the data panel. It can be changed by increasing the number of decimal places globally in Preferences. Otherwise, you can use the 'expression editor' at the far right of the input box (the blue 'f(x)') when changing the value to input smaller values than the FreeCAD number of decimal places.

oliveroxtoby, 2025-06-24:
Hopefully the solution above is sufficient, so I am closing the issue now.