jaheyns/CfdOF #203: Bug reporting - gmsh runs indefinitely, never completes

thandal, 2025-05-08:
For me gmsh doesn't work in FreeCAD CfdOF workbench. All the other meshers work fine, but as everyone knows you want to try all the meshers you can!

What happens is that gmsh just runs and runs and runs. My longest attempt so far is 9 hours for a model the other meshers finish in ~10 minutes.

I dug into things a bit, and found that gmsh was being run with the arguments "-nt 24 -", as in 
[code]gmsh.exe -nt 24 - gmsh/mymesh_solid001_Geometry.geo[/code]

What I found was that the -nt ("number of threads") argument causes gmsh to spin indefinitely, if it is more than 1. I could get it to run just fine on the command line by removing the various arguments, and gmsh completed in ~30 seconds.

I was then able to get it working in the CfdOF workbench by clicking in on the mesh operation in the Model tree, selecting the data tab, and setting the "Number of Threads" to 1 (instead of the default of 0, which automatically tries to use all the threads, 24 in my case).

[image]

I could then write the mesh case, run the mesher, and get a very nice looking mesh!

I don't know if this is an issue with a particular version of gmsh, or with Windows, but I see threads in various forums complaining about gmsh being slow dating back to ~2016 where I *suspect* people have been running into this issue.

oliveroxtoby, 2025-05-08:
This seems to be a bug in gmsh 4.13.1 when running multithreaded on Windows. Please see: https://forum.freecad.org/viewtopic.php?t=87474&start=20#p812814

There's not much we can do but wait for the next release, and run single-threaded in the mean time.

thandal, 2025-05-08:
Understood. I guess the only thing CfdOF could do would be to default to 1 thread on Windows, out of an abundance of caution. But if it's all going to be fixed in the next release anyway, maybe that's not worth the hassle.