jaheyns/CfdOF #204: Running simulations in parallel

Digitalpeer1, 2025-05-22:
Hi,

I am new to CFD and OpenFOAM and I have been running some simulations to get started.
I have noticed my CPU useage to be at 4 cores while running the simulations and was wondering how to use the full 16 threads 8 cores of my 5700x. I have seen that run in parallel being exposed to the gui is in the roadmap but how would I be able to do this manualy?

Thanks in advance!

Lourencosm, 2025-05-23:
Hi,

Generally, using more than four processors when running OpenFOAM on a
desktop computer is not efficient. However, it's also case-sensitive. You
should perform tests with different number of processors.
Go to the system/decomposeParDict file, and change the number of
processors. Decompose the case, run it, compare the calculation times, take
your conclusions.
Also, hyper-threading is not advisable. If you do many OpenFOAM
calculations and use cad software, you should turn hyper-threading off.

Best regards,
Lourenço

oliveroxtoby, 2025-05-23:
Please note that you can simply change the number of processes in the properties of the Solver object and Mesh objects, without the need to manually edit the OpenFOAM case.

For the future, these kind of requests would be more suited to the CfdOF forum (https://forum.freecad.org/viewforum.php?f=37&sid=ba1670bb829341965e32fa407b022e17), as we try to keep this to issue reporting rather than general help requests.