jaheyns/CfdOF #214: Feature request: Speed up install of software deps? cfMesh very slow.

Kheirlb, 2025-07-18:
cfMesh is taking a really long time to "install". I'm >5 minutes in and only 6 MB are downloaded with a really good internet connection. Any chance this can be improved? Also, I think the file is downloaded already, so maybe the text should say "Unzipping" or something.
[image]

Kheirlb, 2025-07-27:
What version of freeCAD are you running? Multiple computers now cfMesh has a slow install. Is there a way I can configure a path like OpenFOAM

Kheirlb, 2025-07-31:
I think I finally figured this one out. You must first specify an installation directly, then OpenFOAM and cfMesh downloads quickly. At least on windows.

oliveroxtoby, 2025-08-25:
Still a mystery to me, and I still can't reproduce the problem :(