jaheyns/CfdOF #217: cfdOF Inoperative; Potential Scripting Issue with v1.31.6

AVTech-Engineering, 2025-08-20:
Attempted to run a routine simulation, getting an immediate error upon initiating the meshing phase of the simulation. We see what appears to be duplicate slashes in the scripting, which is consistent with what the error message is showing as a syntax error. There were no issues with the previous WB version.

Tried to see if it was a HiSA/gmsh currency issue; unable to download any of them; same error showed.

We attempted a uninstall & reinstall of the WB, now dependency checker showing same issue.

Please see attached sceeenshots:

[image]

[image]

[image]

Current FreeCAD Build:

OS: Windows 11 build 26100
Architecture: x86_64
Version: 1.0.2.39319 (Git) Conda
Build type: Release
Branch: (HEAD detached at 1.0.2)
Hash: 256fc7eff3379911ab5daf88e10182c509aa8052
Python 3.11.13, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * boltsfc 2022.11.5
  * Cables 0.2.1
  * CfdOF 1.31.6
  * CurvedShapes 1.0.13
  * Curves 0.6.70
  * Defeaturing 1.2.6
  * DesignSPHysics 0.8.1(29-05-2025)
  * Estimate 0.1.2
  * ExplodedAssembly
  * fasteners 0.5.41
  * fcVM 2024.9.5
  * FeedsAndSpeeds 0.6.0
  * FEMbyGEN 2.5.1
  * FEM_FrontISTR 0.2.0
  * freecad.gears 1.3.0
  * kicadStepUpMod 11.7.3
  * lattice2 1.0.0
  * LCInterlocking
  * MeshRemodel 1.10.38
  * Movie 2025.1.4
  * OpenDark 2024.3.13
  * pcb 6.2023.1
  * Plot 2024.11.26
  * POV-Ray-Rendering
  * Render 2024.12.15
  * sheetmetal 0.7.24
  * Silk 0.2.0

AVTech-Engineering, 2025-08-21:
**Issue solved.**

Spoke with @oliveroxtoby on the forum. Fix was as follows:

Under Edit -> Preferences -> General -> Report View, discovered that "Record Warnings" box was not checked.

Box was checked; restarted FreeCAD. Ran cfdOF Dependency Check, no issues. Ran a simulation without error.

Thank you for the prompt assistance!

oliveroxtoby, 2025-08-21:
Strange! I didn't expect that to have any influence on the result - just to give more info on what was going wrong. Anyway, glad you got it sorted!

closed 2025-08-21 by oliveroxtoby