jaheyns/CfdOF #218: descussion: tring to add the ability to use AMI/NCC

muezabdalla, 2025-08-24:
I am trying to add the ability to use AMI/NCC depending on the version of OF but to set the boundary condition 
I need to know which version is used. for example for U it will be like this:
```
// for AMI
patch_name
{
    type cyclicAMI;
    value   uniform (0 0 0);
}

// for NCC
patch_name
{
    type        movingWallSlipVelocity;
    value       $internalField;
}
```
one clunky way to fix this is to let the user choose in the GUI between AMI and NCC and process depending on that decition
or if it is possible to define the boundary condition base on the type of the patch that would be the best solution but I did not find a way to do that

so can anyone help with this. and thanks in advance

oliveroxtoby, 2025-08-25:
I like to write the same case files that will be portable across different OpenFOAM versions and have any version-specific selections be coded into the Allrun or Allmesh scripts. There are many examples, such as the following in the Allrun file:
```
%{%(dynamicMeshEnabled%)
%:True
mkdir -p 0/include
if [ -z ${FOAM_API+x} ] && [ "$WM_PROJECT_VERSION" -ge 9 ]
then
    echo "type internal;" > 0/include/helperPatchFieldType
    echo "type internal;" > system/helperPatchType
else
    echo "type fixedValue;" > 0/include/helperPatchFieldType
    echo "type patch;" > system/helperPatchType
fi

%}
```
This writes different things into an include file called `0/include/helperPatchFieldType` file for Foundation OpenFOAM >= v9, and this include file is included from the `0/_boundary_redistributeHelper` file which in turn written into each `0/<field>` file.

I think you'll want to do something very similar to this.

muezabdalla, 2025-08-25:
ok thank you.

closed 2025-08-25 by muezabdalla