jaheyns/CfdOF #225: Bug reporting? CFDoF Initialise zones for multiphase "boat" on freesurface - velocity = 0

hayeseed, 2025-09-09:
Hi, I'm learning CFDoF, done quite a few of Tech Bernd's Utube Tutorials.

I'm trying to work out how to roughly replicate wigleyHull, a "boat" @ constant velocity water.
I have tried many combinations of Initial conditions, initialise zones etc.  This is my best attempt yet.

The problem is that the flow has no velocity.  Yet I have set V=1m/s in many dialogs.
Wierdly, the hull surface shows velocty of 1mm/s, which is suposed to be the flow Velocity.

To get multiphase working I have defined 2 initialisation zones, water below z=0 and air above.
Also set up inlets and outlets for both air and water phases.
Before setting up separate inlets and outlets I got very random results.
Muliphase now looks good in paraview.

Model attached.
It meshes in seconds and runs in a minute or two.

I would LOVE some feedback on what initial conditions and boundary conditions would get reasonable results.
Is it CFDoF?  or is it me?  :)

[image]

[WigleyHullDave2.zip](https://github.com/user-attachments/files/22224877/WigleyHullDave2.zip)

muezabdalla, 2025-09-16:
Hi @hayeseed sorry for being late but here is what I found

the inlets and outlets are set on another body so they did not appear on the simulation
you can fix that by making 2 pads from sketch `Sketch` one pad is 500 reversed and the other is 100 and set the `refine` on the pads and the Cut to false

and you will have to set the symm, inlet_air, inlet_water, outlet_air and outlet_water boundary again on faces that are on the `cut`  not `AirVol` nor `WaterVol`

here is the fixed file but now it will take hours to run (maybe because of the fast speed)

[WigleyHullDave3.zip](https://github.com/user-attachments/files/22369820/WigleyHullDave3.zip)

hayeseed, 2025-09-17:
@muezabdalla , thank you, thank you, thank you.
Nice little technique there to turn off refine and hence get separate surfaces on the "Cut" part to apply different conditions to air and water.  
I never would have found that one :)

After running that model, here are the results...

[image]

[image]

[image]

Looks to me like the Velocity of air at the outlet is > 400m/s !

What am I missing?
Is there are better "Standard" approach to analysis of boat on water in CFDoF?

I'm also curious about your comment about "the fast speed".  I set the velocity to 1m/s over a 2m model.  It's only about 2 knots which is really slow for a boat.  Have I got something wrong (maybe with units) ?

muezabdalla, 2025-09-17:
>Looks to me like the Velocity of air at the outlet is > 400m/s !

that maybe because of the open boundary condition

 >Is there are better "Standard" approach to analysis of boat on water in CFDoF?

I do not think so.

>I'm also curious about your comment about "the fast speed". I set the velocity to 1m/s over a 2m model. It's only about 2 knots which is really slow for a boat. Have I got something wrong (maybe with units) ?

generally when you increase the velocity that increase the courant number which will decreace the deltaT which will increase the simulation time. in other words
more speed = more courant number = less deltaT = more simulation time

on my computer the deltaT is about 8e-10 so I thought that decreasing the velocity might help but looks like you are not facing this problem in the first place because you got the results very fast

hayeseed, 2025-09-30:
Thanks @muezabdalla,
Here's a summary of what i learned...

- Create the CFD volume using one body, split into 2 pads
- To achieve this, 2 Pad feature for Water and Air were used, with the 2nd one requiring the Refine setting in the data table set to false.  -  Part Design, Refine = false.
- Create Boat Geometry via whatever method and subtract it from the CFD body.
- All inlets, outlets, walls, symmetry etc need to be applied to the same CFD body.
- This created separate faces for inlet air, inlet water, outlet air and outlet water.

- Initialise Fields set Alpha.Water to 1

- Create a separate body for the initial Air volume above the water surface.
- Initialise Zone set Alpha.Water to 0 and select the separate "air" body.

Here's my other settings...

- Physics Model = Transient, Multiphase, Laminar or RANS, set gravity direction and acceleration.
- Initialise Fields = Velocity from Inlet_water boundary, Pressure = 0 Pa in my case (need to explore this more, Volume Fraction = Water, 1
- Initialise Zone for Air = Air Solid selected, Velocity = V = 1m/s, Volume Fraction = Water, 0.
- Inlet Water = Uniform Velocity, Select water inlet face, V = 1m/s, P = 0Pa, Volume Fraction = Water, 1
- Inlet Air = Uniform Velocity, Select air inlet face, V = 1m/s, P = 0Pa, Volume Fraction = Water, 0 
- Outlet Water = Static Pressure, Select water outlet face, V = 1m/s, P = 0Pa, Volume Fraction = Water, 1
- Outlet Air = Static Pressure, Select air outlet face, V = 1m/s, P = 0Pa, Volume Fraction = Water, 0 
- Constraint Symmetry for CFD body sides and bottom
- Open Top Face = Ambient pressure, P = 0Pa, Volume Fraction = Water, 0 
- Boat Hull = Wa;;, No-Slip, select all faces of hull including those above the water surface.

Thanks again for the help.
I would happily take advice on any improvements to this method.
Cheers,
Dave